AlkaMotors/AM32-MultiRotor-ESC-firmware #6: Feature request: confimation beep on power-up

DzikuVx, 2020-10-25:
Currently Am32 beeps motors only after it detects valid input from FC. This is confusing. 
Generate short series of beeps right after power is connected and MCUs booted to inform the user that power is delivered

Voodoobrew101, 2020-10-25:
It beeps until it is armed. In biderectional mode forward begins right at 1000. I have only been studying bidirectional mode. The arming beep is a different sound. Are you getting 2 sounding beeps? I also feel it has issues arming in a crawler though. I actually found turning it on with high throttle, then lowering the throttle till it arms works most reliably. Also i dont have the issue with all of my pwm sources.

AlkaMotors, 2020-10-25:
Hi, thanks for the feedback, this has been fixed with the lastest version of the bootloader. Right now the issue is without a signal at all the bootloader doesn't jump to the main code. It would require a reflash using the swd pads but the latest versions will beep on power up and no signal.

AlkaMotors, 2020-10-25:
@DzikuVx your esc must have the old bootloader on it still.

DzikuVx, 2020-10-25:
Perhaps. Is there a simple way to flash BL? I should have STlink somewhere

JyeSmith, 2020-10-25:
St link and the st link utility is the easiest way.  I mounted the ESC upsidedown purely so I had easy access for any BL changes.

AlkaMotors, 2020-10-26:
Yeah, the st-link utility or cubeprogrammer from st will do it. For the ifligth target it's the pb4 bootloader. The version on GitHub (v4) doesn't have any changes apart from that startup beeps so it's safe to keep using the old one.  There will be a new version v5 on there tonight that adds compatibility with older f3 flight controllers too in case you want to wait to update to that one. It's not critical though the old bootloader will always be able to flash and change settings still.

DusKing1, 2020-10-26:
On the back side they have SWD pads.

AlkaMotors, 2020-10-27:
@DzikuVx bootloader source is now here https://github.com/AlkaMotors/AM32_Bootloader_F051. There is two new binaries in the release folder in that repo. Not critical though but v5 works with older FC's too.

AlkaMotors, 2020-10-30:
I am going to close this issue as it is fixed with the new bootloader. If you have any issues with it let me know.

closed 2020-10-30 by AlkaMotors