AlkaMotors/AM32-MultiRotor-ESC-firmware #9: Feature Request: Linux Configurator

stronnag, 2020-10-28:
Please might we have a Linux configurator (or a source release so it can be ported).

AlkaMotors, 2020-10-29:
Hey stronnag, I have a linux version already but I have a couple of finishing touches to do. This weekend I will get it uploaded into the release folder. Source will be up soon too. There is also work being done to add compatibility with blheli_configurator that will be coming in the future.

stronnag, 2020-10-29:
Thanks. Look forward to being able to test it.

AlkaMotors, 2020-10-31:
Check it out and see if it works for you.
https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/tree/master/Release/CONFIG%20TOOL

stronnag, 2020-10-31:
Tested (Arch Linux). No problems. 
Thanks. Look forward to be able to build it so it can use local preferences (like qt5ct).

closed 2020-10-31 by stronnag