AlkaMotors/AM32-MultiRotor-ESC-firmware #18: Feature request: eprom define the names and parameters for the configurator

Voodoobrew101, 2020-10-31:
Because the crawler and quad parameters are drasticaly different. This way you donvt need to update the configurator as often also. Then configure the eprom with some #define yada ya, so it contains the name of the value, the possible values, the default and/or current value, and what type of input box (type or slider). 
Just an idea not sure if the eprom has room for it.

AlkaMotors, 2020-11-13:
We do have a lot of space for EEprom settings. The configurator will still have to know something about the data it is receiving to brake it down into its parts but I think a separate space for crawlers might be a good plan. We are currently using 48 of 1024 Bytes in flash space for settings. Its not a bad idea to have the eeprom define the settings parameters but it does make the size of the data transfer much larger each time.

Voodoobrew101, 2021-07-21:
In order upload decently descriptive names for the sliders It does seem like it would be rather wasteful.
[image]

closed 2021-07-21 by Voodoobrew101