AlkaMotors/AM32-MultiRotor-ESC-firmware #22: Pin mapping.

Marsma13, 2020-11-09:
I have to ESC with fried MCU. Pyrodrone and Airbot. I know that Pyrodrone use F350 MCU but it is pin compatible with F051. I can prepare pinout for it but i have to know what exactly you need to know for being able to build FW for it. Can i help you somehow?

AlkaMotors, 2020-11-09:
What i would need to know are the 6 pins to the MOSFETs driver and the associated 3 comparator pins back from voltage dividers from the motor phases. Then which pin is the input and telemetry These are mostly the same between targets, it's just the order that changes. Take a look at the targets file and one mighty match yours already ..the mp631 matches the wraith32 target.https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/blob/master/Inc/targets.h

Marsma13, 2020-11-09:
Ok so i take multimeter in hand and do a measurement. 

If the FD6288 have proper mapping situation is like this:
LIN1 - PB1
HIN1 - PA10
DIV1 - PA0
LIN2 - PB0
HIN2 - PA9
DIV2 - PA4
LIN3 - PA7
HIN3 - PA8
DIV3 - PA5

TELE - PB6
INPUT - PA2

COMMON to all phases to one resistor - PA1

As I see it is almost the same as FD6288 from configs just phases are swapped out

I marked bold as the same and italic as difference. 

Cannot see the telemetry input and common input as it may be the timers, but it is not clear to me. 

Above mapping is for 
Airbot Ori32 BLHeli32

I will do mesurements for Pyrodrone ESC tommorow.

#define FIRMWARE_NAME           "FD6288_PA2  "
#define DEAD_TIME               45
#define USE_TIMER_15_CHANNEL_1
**#define INPUT_PIN               LL_GPIO_PIN_2
#define INPUT_PIN_PORT          GPIOA**
#define IC_TIMER_CHANNEL        LL_TIM_CHANNEL_CH1
#define IC_TIMER_REGISTER       TIM15
#define IC_TIMER_POINTER        htim15
#define INPUT_DMA_CHANNEL       LL_DMA_CHANNEL_5
#define DMA_HANDLE_TYPE_DEF     hdma_tim15_ch1
#define IC_DMA_IRQ_NAME         DMA1_Channel4_5_IRQn

**#define PHASE_A_GPIO_LOW        LL_GPIO_PIN_1
#define PHASE_A_GPIO_PORT_LOW   GPIOB
#define PHASE_A_GPIO_HIGH       LL_GPIO_PIN_10
#define PHASE_A_GPIO_PORT_HIGH  GPIOA**

**#define PHASE_B_GPIO_LOW        LL_GPIO_PIN_0
#define PHASE_B_GPIO_PORT_LOW   GPIOB
#define PHASE_B_GPIO_HIGH       LL_GPIO_PIN_9
#define PHASE_B_GPIO_PORT_HIGH  GPIOA**

**#define PHASE_C_GPIO_LOW        LL_GPIO_PIN_7
#define PHASE_C_GPIO_PORT_LOW   GPIOA
#define PHASE_C_GPIO_HIGH       LL_GPIO_PIN_8
#define PHASE_C_GPIO_PORT_HIGH  GPIOA**

_#define PHASE_A_COMP COMP_PA5_
**#define PHASE_B_COMP COMP_PA4**
_#define PHASE_C_COMP COMP_PA0_

Marsma13, 2020-11-09:
For Pyrodrone F3 34A ESC mapping is the same as TMOTOR 55A
INPUT is PB4
TELE is PB6
COMMON is PA1

And phases are:
LIN1 - PB1
HIN1 - PA10
DIV1 - PA0

LIN2 - PB0
HIN2 - PA9
DIV 2 - PA4

LIN3 - PA7
HIN3 - PA8
DIV3 - PA5

AlkaMotors, 2020-11-09:
Great, that is just the info we need. So far all targets use pb6 for telemetry and pa1 is common to all for the comparator virtual neutral so it's not in the config file. That's also the same as iflights pinout. The tmotor just has an extra led function.

AlkaMotors, 2021-02-10:
Added to target compatibility list in wiki.

closed 2021-02-10 by AlkaMotors