AlkaMotors/AM32-MultiRotor-ESC-firmware #37: Add new Dshot commands to configure ESC settings.

Quick-Flash, 2021-03-27:
The usefulness of changing ESC settings is becoming more and more common knowledge. The ability to change esc settings using the FC configurator or OSD will make testing setting changes far easier. Working towards new Dshot commands that will work for AM32, Bl32, KISS, blheli_s (through bluejay), and APD setup of various settings will improve the user experience for many.

Zuldan, 2021-03-27:
Imagine being able to configure ESC settings in your OSD while trying to tune your quad :-) I could finally leave the laptop at home!

mikeller, 2021-03-28:
One idea that was brought up in a discussion, and that might be feasible with the relatively limited facilities of Dshot commands would be to assign a couple of Dshot commands to activate a number of 'preset' configuration.
This way, these commands could be used to switch between individual configurations without having to know what the parameters supported by the firmware used are.

@ronlix, @sskaug, @tonybazz, @mathiasvr: What are your thoughts on this?

stylesuxx, 2021-03-28:
Another idea was to have a general "Payload" Dshot command to send arbitrary data, basically having a "sub-protocol" spanning multiple "Payload" command frames.

Zuldan, 2021-03-28:
Another idea... You could adjust your ESC Rampup power depending on the lipo you plug by linking this to profiles based on lipo cell count. For example if you have 4S motors and you decide to use a 6S battery, you could set the 6 cell profile to adjust the ESC Rampup Power to 25% and then back to 50% Rampup when 4S is plugged in. So many different options!

Quick-Flash, 2021-03-28:
> Another idea... You could adjust your ESC Rampup power depending on the lipo you plug by linking this to profiles based on lipo cell count. For example if you have 4S motors and you decide to use a [...]

this could be done on the fc side once the necessary dshot commands are added.

saidinesh5, 2021-03-28:
A DShot command I wanted was one that lets you specify the beacon tone, duration, and maybe strength.

This would
* Let the flight controller play more unique melodies when your quad crashes
* Allow for interesting error/warning tones than the hardcoded ones.

ronlix, 2021-03-28:
> One idea that was brought up in a discussion, and that might be feasible with the relatively limited facilities of Dshot commands would be to assign a couple of Dshot commands to activate a number [...]

Hi,

im fine with adding presets. as far as i know (im not totally aware of what other FWs have added) there should be some dshot commands free for now that can be used. 

> Another idea... You could adjust your ESC Rampup power depending on the lipo you plug by linking this to profiles based on lipo cell count. For example if you have 4S motors and you decide to use a [...]

using the ESC to scale down the bettery voltage is quite inefficent. ita always better to use the right motor KV or battery voltage.

mikeller, 2021-03-29:
@ronlix: For what I know these are all the Dshot commands that were either defined in collaboration or claimed unilaterally by individual manufacturers: https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/src/main/drivers/dshot_command.h#L34-L61

4712, 2021-03-29:
What is actually meant with presets?
Should this be user editable sets of all values in the configuration?
This could get quite complex for BLHeli_32, because the bootloader in combination with the maximum payload of the 4way-interface restricts writing to the settings area 256 byte minus overhead.
What about just using an android device in the field to optimize the settings with the app?

Quick-Flash, 2021-03-29:
> @ronlix: For what I know these are all the Dshot commands that were either defined in collaboration or claimed unilaterally by individual manufacturers: https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/src/main/drivers/dshot_command.h#L34-L61

If i'm reading this right we have space for 16 more commands, unless we remove some of the unused commands. 

Do these two commands `DSHOT_CMD_SETTINGS_REQUEST, // Currently not implemented
DSHOT_CMD_SAVE_SETTINGS,` allow us to view the esc settings and tell the esc to save setting changes?

mikeller, 2021-03-30:
@4712:

> Should this be user editable sets of all values in the configuration?

Ideally, but not necessarily - I don't think that it will be useful for users to use different presets for things like beacon settings.

> This could get quite complex for BLHeli_32, because the bootloader in combination with the maximum payload of the 4way-interface restricts writing to the settings area 256 byte minus overhead.

Could this be circumvented by having one command only write the configuration for _one_ preset, with the preset number being part of the configuration that is written?

@Quick-Flash:

> Do these two commands `DSHOT_CMD_SETTINGS_REQUEST`, `DSHOT_CMD_SAVE_SETTINGS`, allow us to view the esc settings and tell the esc to save setting changes?

Almost. `DSHOT_CMD_SETTINGS_REQUEST` is currently not implemented by any ESC firmware (that I know of), as all of the settings (and other data, like the ESC type / firmware version) are returned by `DSHOT_CMD_ESC_INFO`. So yes, this can be done, but it requires a separate telemetry UART connection to read the data sent by the ESC.

Quick-Flash, 2021-04-11:
Is this something that we are still going to try and make happen?

closed 2025-01-20 by Quick-Flash