AlkaMotors/AM32-MultiRotor-ESC-firmware #46: BF 4.3 (10.8) Motor Rotation dialog does not fully work with AM32

limonspb, 2021-06-25:
Motor Rotation dialogue (From Betaflight configurator 10.8) does work with BLHeli_32, BLHeli_S and JESC, both versions Vizard and Individual buttons.

However, with AM32 the behaviour is not so nice.
Every time when FC sends a command to change motor rotation the AM32 restarts (can hear tones) and sometimes does not start spinning again.
I guess this is happening because when AM32 saves settings it also restarts the whole ESC.
Wondering if there is a way to do things similar to Blheli_S here?

Thanks!

AlkaMotors, 2021-07-05:
Hi limonspb. Try the version 1.72 release. This was fixed a while ago in the experimental branch. Hopefully it still is.

limonspb, 2021-07-21:
Latest version 1.74 works like a charm with the dialog. Closing the issue.

closed 2021-07-21 by limonspb