AlkaMotors/AM32-MultiRotor-ESC-firmware #57: iFlight ESC flashing

benedikt-bartscher, 2021-08-04:
Hello there, thank you for this great alternative to BLHeli! I hope it will replace BLHeli soon :)

I wanna flash my iFlight 4in1 ESCs with AM32. I already bought a STLinkv2 and tried to connect with CubeProgrammer but when i press "Connect" nothing happens. I assume wrong wiring. I powered the ESC via its main DC input. There are 4 pads next to each ESC on the board, i hope it's 3.3V, GND, SWD and SWC. I definitly know one pin is 3.3V and one GND. However connecting the SWD and SWC to the last 2 pins doesnt work in either direction..

Did anyone flash a iFlight 4in1 before? Are these pins SWD and SWC?

Thanks in advice!

AlkaMotors, 2021-08-05:
A ground must also be connected from the st-link to the mcu. Not sure if cube programmer is still going to complain since that's not a stm32 processor.  The picture shows an mcu made by china key systems, the st-link utility might be better to flash a cloned mcu.  Please Join the slack for help or st-micro forums.. this really shouldn't be opened as an issue for am32. We have no control over the st-link and their software.

benedikt-bartscher, 2021-08-06:
Of course i connected Ground between ST-Link and the board. Thank you, i will try out STLink Utility this evening! You are right, this isn't an AM32 issue. Wasnt aware of the slack channel - i will close this issue.

closed 2021-08-06 by benedikt-bartscher