AlkaMotors/AM32-MultiRotor-ESC-firmware #65: Changing sine power

Voodoobrew101, 2021-08-29:
You chose to change the denominator instead of the numerator so this number seems kinda backwards. 
I tried setting the denominator to 10 for many of my builds and didn't have any problems. 
This lets the power range be sat as 1-10.
1-100 would be preferable but I dont know if that would make the numbers too high.

Voodoobrew101, 2021-08-30:
I would change this 
`		TIM1->CCR1 = ((2*pwmSin[phase_A_position]/SINE_DIVIDER)+gate_drive_offset)*TIM1_AUTORELOAD/2000;
		TIM1->CCR2 = ((2*pwmSin[phase_B_position]/SINE_DIVIDER)+gate_drive_offset)*TIM1_AUTORELOAD/2000;
		TIM1->CCR3 = ((2*pwmSin[phase_C_position]/SINE_DIVIDER)+gate_drive_offset)*TIM1_AUTORELOAD/2000;`

to 
`		TIM1->CCR1 = ((Sine_Power*pwmSin[phase_A_position]/100)+gate_drive_offset)*TIM1_AUTORELOAD/2000;
		TIM1->CCR2 = ((Sine_Power*pwmSin[phase_B_position]/100)+gate_drive_offset)*TIM1_AUTORELOAD/2000;
		TIM1->CCR3 = ((Sine_Power*pwmSin[phase_C_position]/100)+gate_drive_offset)*TIM1_AUTORELOAD/2000;`

This would make sine power be a percentage 1-100 and the default would be 66
I know this might be a dangerous setting to give to people.
But I have had to change it for use in my scx24 where I was using a tiny 1408 motor and at 50% power (what I used in my scx10 with larger motors) the motor wouldn't run in sine mode it just was erratic. Raising the power to 100% fixed the problem. 
So maybe to a certain extent this setting should be more modafiable?

AlkaMotors, 2021-09-01:
I agree we need some type of adjustment but we have be careful.. I'll look into adding a range. Even just a 3 step low medium and high would probably even be enough.

Voodoobrew101, 2021-09-04:
I don't know if I would put it in the configurator or not though, If maybe you make people type "I will monitor thermals" before letting them change it, because obviously it can cause damage.

Voodoobrew101, 2022-01-11:
Now that this is in the configurator there is no need to change this setting

closed 2022-01-11 by Voodoobrew101