AlkaMotors/AM32-MultiRotor-ESC-firmware #67: T-Motor F45 single ESC

blitz2190, 2021-09-03:
So I am trying to get this working on a T-Motor F45a single ESC, Before flashing I did test blheli and all was working with the motor. I was able to flash PA2 version 10 bootloader and the latest firmware as well as set the settings, It powers up beeps and arms, but when trying to move the motor it cogs terribly and gets hot very quickly. I have tried every combination of the 3 wires to the motor with no success, I can tell which combo is correct by the clear tone when starting up. Any Ideas?

Voodoobrew101, 2021-09-04:
What firmware are you using?
Specifically what target? 
Try different targets until it runs better. The t-motor 45a target is for the 4in1 , totally different design.

The combination of the 3 motor wires has nothing to do with this. they only control direction. (swapping any 2 will reverse the motor direction.)

blitz2190, 2021-09-04:
so the AM32_FLYCOLOR, FD6288, HGLRC, MP6531, SISKIN all seem to work, as in no heat and smooth running once moving, the low speed startup seems ok but right between low speed and about 1/4 throttle it cogs, then its smooth up to, even low speed seems a bit off but its not noticably cogging

Voodoobrew101, 2021-09-05:
You are 100% correct. 
Sine range is like a throttle curve. 
Startup power may smooth it out.
But it is expected behavior unfortunately. 

Occasionally tweaks are added that improve the transition, but it is not perfect.

blitz2190, 2021-09-05:
issue was mostly fixed by playing around with the settings and tuning, thank you for the help

closed 2021-09-05 by blitz2190

Voodoobrew101, 2021-09-05:
I don't understand how the different targets work, but some of those might cause sine and normal running to be in opposite directions.