AlkaMotors/AM32-MultiRotor-ESC-firmware #69: RC Car mode

abrarakram19, 2021-09-10:
Hi bro, I am interested in developing ESC hardware for RC car and want to use your code for initial prototypes but I am not good in coding. I want to assign pinout on my choice for better layout and functionality. I already designed layout and tested some other open source codes like blheli and simonK on AVR 8bit microcontrollers and learned Assembly to make Some basic modifications but I am still not good to make some more modifications to add brake function before reverse throttle. How you can help me? Or what you will suggest. Can anyone help me to make some modifications in above mentioned Assembly code? Because it will be more easy for me to implement those codes for my existing layout design. Any type of help will be highly appreciated.

AlkaMotors, 2021-09-10:
All you need to do is enable RC car mode in the config tool. First pull of reverse applies a brake. Double tap to reverse. https://www.youtube.com/watch?v=vwFFFg4DxZs&list=UUvq7C-yEi_syF4zkxDDkKGQ&index=2
The pinout of your choice has to be the ones that are connected to the proper peripherals! There is not a lot of flexibility.  We have a targets file that defines these pinouts. https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/blob/master/Inc/targets.h

abrarakram19, 2021-09-11:
Thank bro, will code work on f051c8k6 same way? And can I use 3 high and
low side drivers instead of 6288 or 6531 in my layout?

abrarakram19, 2021-09-11:
Sorry I mean f051k6t6? And drivers ir2101s?

closed 2023-03-08 by AlkaMotors