AlkaMotors/AM32-MultiRotor-ESC-firmware #70: ESC startup melody?

howels, 2021-09-26:
Noticed reference to startup melody in https://github.com/stylesuxx/esc-configurator/issues/108#issuecomment-829146025 and wondered if this is possible yet?

Voodoobrew101, 2021-09-28:
Not yet, but it is possible to change the startup sounds to pretty much any tone If you edit the sounds.c
Thats how some of mine play Dixie
Then you compile the firmware,
its easiest to use the actions tab here in github to build the firmware IMO. 
or you can use visual studio code with WSL installed.

howels, 2021-09-29:
> Not yet, but it is possible to change the startup sounds to pretty much any tone If you edit the sounds.c Thats how some of mine play Dixie Then you compile the firmware, its easiest to use the [...]

I notice in `sounds.c` there is a reference to "bluejaytune" which reads a tune from EEPROM and plays it if found.  How can we write that tune? Would prefer to experiment with tunes using esc-configurator.com instead of re-compiling each time.

howels, 2021-09-29:
Is this a "wait for the next release" item? Seems that the code is in master and esc-configurator.com is waiting to include AM32 melody support using the same routines that bluejay uses.

AlkaMotors, 2021-11-01:
@howels am32 version 1.78 and the online esc configurator are now working for custom melodies.

closed 2021-11-01 by AlkaMotors

howels, 2021-11-01:
> @howels am32 version 1.78 and the online esc configurator are now working for custom melodies. https://esc-configurator.com/

thanks for letting me know, I think I got a bad flash trying to use an earlier version with esc-configurator but I'll re-flash via ST-Linkv2 soon as I need to strip the car down again.