AlkaMotors/AM32-MultiRotor-ESC-firmware #76: bl esc on slot car

mtaddei, 2022-03-07:
Gd day from Italy,
we are a group of slot car modeller that are trying to fit brushless motor on slot car.
We'll like to know if it is possible turn off the radio pwm signal setting it always at max speed
enabling us to controll the motor speed directly with the voltage supplied by our slot controller
on the track braid

Tnks for your kind replay

Max

AlkaMotors, 2022-03-18:
Yes, there is a mode made just for this purpose. You just need to compile the firmware with those options enabled. If you have a specific esc in mind i can attach one here.

mtaddei, 2022-03-18:
the esc is FEICHAO 40 AMP

mtaddei, 2022-03-18:
just removed the wrapping and the processor is an ATMEGA8AU, is compatible or not?

AlkaMotors, 2022-03-21:
No, 8 bit mcu's are not supported. It must be a 32 bit arm mcu.

AlkaMotors, 2022-03-23:
No 8 bit mcu support.

closed 2022-03-23 by AlkaMotors