AlkaMotors/AM32-MultiRotor-ESC-firmware #79: No power motor

guchiboy, 2022-03-31:
Hi! I installed AM32_IFLIGHT_1.88 On ESC FLYCOLOR 50A. After switching on, the motor emits a signal, but does not respond to pressing the gas trigger. HolmesHobbies revolver s40 1800kv motor. All I did was unlock the MCU. And then I flashed only the hex file. SerialPortConnector connects to esc, the settings are visible. Can you tell me what could be the matter?

AlkaMotors, 2022-04-02:
Did you turn on bi-direction in settings? If it's making a chime on startup then the flashing worked. If bidirection is on and your radio is putting out close to 1500us neutral then it will have a startup chime. Then you will here an arming chime. Then you can drive.

AlkaMotors, 2023-03-08:
since there has been no response for a year i will assume you managed to arm.. closing

closed 2023-03-08 by AlkaMotors