AlkaMotors/AM32-MultiRotor-ESC-firmware #80: No TEKKO32 target

negre, 2022-04-07:
hi,
The Tekko32_4in1 is listed in the supported harware wiki page, but the target "TEKKO32" is not referenced in the code and there is no binary generated with this name.

AlkaMotors, 2022-04-12:
There should be no TEkko32 in the wiki page either. I have never made a target for it. I will edit that page.

AlkaMotors, 2022-04-12:
same issue with EMAX. there is no target for that in the code either. One of the drawbacks of having a publicly editable wiki. Before I delete them maybe @Viktor055 can fix them up?

Viktor055, 2022-04-13:
Hi guys. Yes, I managed to repair my Tekko32. I changed the firmware for my ESC, there was a difference in the ABC phases

AlkaMotors, 2022-04-13:
@Viktor055 that's good!  but there is no target added to the code for it. Can you also do a PR with your changes to the targets.h file ? Otherwise people trying to flash a tekko will not have a firmware.

Viktor055, 2022-04-13:
> 

Hi. I'm sorry, but I'm a noob in github)) I was able to copy the entire branch to myself and made changes to myself. Maybe you can copy it from me?

AlkaMotors, 2022-04-13:
ok, i see it in you github repo. Hardware group E was retired as it is the same as hardware group A. @Viktor055 the FD6288 target should work on the Tekko. If possible for you to check that then i can make a separate TEKKO32 target.

Viktor055, 2022-04-13:
Ok. I will check it out

Viktor055, 2022-04-16:
Yes it works on Tekko32. Today I flashed the FD6288 controller and the ESC successfully started the motor. But there is one feature with this firmware, it is a large motor speed. More than 1800 there is already a synchronization failure due to high speeds. It was also on my firmware. I couldn't limit it with settings.

AlkaMotors, 2022-04-17:
@Viktor055 You will have to give more details. 1800 what? kV? Throttle? What kind of motor? Firmware version? What are your settings. Did you build the firmware yourself or use version 1.88 release?

Viktor055, 2022-04-17:
1800 Throttle
motor 2216 900KV
Firmware AM32_FD6288_1.88
I also checked it on the assembled firmware - and it was the same - a high rotation speed. I could not give more than 1800 because there were failures in the motor, it was due to high speed.
That is, I want to say that at a speed of 1800 Throttle, the motor turns at maximum speed, which is not typical for it.
I tried all the settings that are in the configurator Multi ESC Config Tool 1.82 and could not limit the rotation speed.

I have a Tekko32 with the original firmware. I tested it with this motor 2216 900KV. And then the motor does not gain such great momentum.

AlkaMotors, 2022-04-17:
At 900kv you are no where even close to the maximum rpm of the firmware it can handle. So if I understand with the original firmware the motor does not rotate that fast. But with AM32 it rotates faster but fails at 1800 throttle? It sounds like there is something else wrong. There is no way to limit rotation speed in either AM32 or Blheli32 firmware ..the esc should be running the motor as fast as it can.

Viktor055, 2022-04-17:
In the original Tekko 32 firmware from Blheli32, the motor operates in the entire range from 1000 to 2000 Throttle.
And reaches full rotation speed at 2000 Throttle.
In the firmware AM32_FD6288_1.88, the motor runs from 1000 to 1765 Throttle.
And when you add 1800, the motor goes beyond its capabilities and failures begin.
Unfortunately, now I have nothing with which to measure the real speed of the motors. But it's faster than usual.

AlkaMotors, 2022-04-17:
AM32 limits for a 14 pole motor would be about 60,000 rpm.for a 900kv motor this would be about 60v where you start having trouble. If the motor spins faster than the rated kV then there is some sort of issue, either a bad coil or some issue with the esc hardware or just the wrong target entirely. There has never been any difference in speed between AM32 and Blheli32.

Viktor055, 2022-04-17:
I'll find a device to measure the rotation of the motor and make measurements. Perhaps this is due to my 9-pole and 900KV motor

PoseidonFPV, 2022-05-05:
Hi huys if anybody solve the issue ? I have just bought tekko32 f4 45a http://www.holybro.com/product/tekko32-f4-45a-esc/
 single esc and i am affraid now that i will not be able to change firmware to Am32 :( please help me, what to do ?

Viktor055, 2022-05-06:
Your controllers use a new microcontroller F4. There is no bootloader and AM32 firmware on it.
But perhaps the author will do it in the future.

PoseidonFPV, 2022-05-06:
Thank You for reply :) hope they will :P last night i have ordered emax
formula 45a this one should do the job :)

Be in touch :)
Pawel

AlkaMotors, 2022-05-06:
@PoseidonFPV as soon as I get my hands on one of the f4 MCUs I will try and add support for it. I just haven't actually had one on the bench yet.

PoseidonFPV, 2022-05-06:
Get it from aliexpress tekko32 f4 is really cheap :)

US $22.13 | Holybro Tekko32 F4 MCU BLHeli32 45A ESC Dshot1200 2 ~ 6S RGB
LED dla dronów wyścigowych FPV
https://a.aliexpress.com/_mrl1qGa

Thanks for reply ;)
Pawel

nzmaniac2, 2022-07-01:
hi everyone did the  Tekko32 F4 MCU get added im also looking at the tekko chip

AlkaMotors, 2023-03-08:
f4 added as of 1.93

closed 2023-03-08 by AlkaMotors