AlkaMotors/AM32-MultiRotor-ESC-firmware #83: Dose MCU:ARTERY-AT32F421K8U7 be support?

JamesKlark, 2022-05-14:
I bought a HAKRC 35A,but when i received it ifound they changed the MCU to ARTERY-AT32F421K8U7.I wonder if this MCU be support?

AlkaMotors, 2022-11-22:
@JamesKlark artery has support now but the flashing is still the problem. If you have an at-link i can give you some test files.

howels, 2022-11-23:
> @JamesKlark artery has support now but the flashing is still the problem. If you have an at-link i can give you some test files.

Could we please get a (brief) wiki page outlining the process of flashing from blheli32 to AM32 on ArteryTek hardware with an at-link?

adie0986, 2023-02-07:
Based on Alka Motors YouTube channel, I do prepare a step-by-step guideline. I believe that most already have ST-Link and can use that to flash the Artery Tek. Hope this will help... 

[Step to flash AM32 to Artery Tek MCU using ST-Link.pdf](https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/files/10676469/Step.to.flash.AM32.to.Artery.Tek.MCU.using.ST-Link.pdf)