AlkaMotors/AM32-MultiRotor-ESC-firmware #87: At 1733 throttle, the motor stalls

AkunEverson, 2022-08-08:
Hello!
During debugging, the following two problems occurred:
1. When the throttle is pushed to about 1730, click to stop;
2. With the acceleration of the motor speed, the voltage value at the input end also rises: 48V rises to 53V (at this moment, the  throttle: 1730, the speed r / min: 3212) and stops;
The test conditions are:
12s battery
Install the blades
Motor: 110KV, 42 poles
Supplement:
When the blade is not installed, the above two problems do not occur.

AlkaMotors, 2022-08-08:
I am not sure i follow this. So with a prop on when it gets to a certain RPM the esc is shutting off? is it resetting? Is there any way you can do a video clip?

AkunEverson, 2022-08-09:
First of all, thank you for your reply!
I recorded a video about these two phenomena. You can watch it. As for the ESC circuit board, I designed it myself according to your program.
https://user-images.githubusercontent.com/110520763/183555989-f57e1e12-a0c6-4e1f-8950-c228006ace44.mp4
As for the phenomenon that the motor stops rotating, I consider whether there are any protection measures in the program, or the speed limit is reached, which causes the motor to stop rotating. According to the phenomenon in the video, you can analyze it. Thank you very much.
Note: in the video, the remote red digital tube number: PWM generator (throttle value), and the data on the nearby instrument (power supply device) are: current, voltage, pushing force, torque and rpm.

AlkaMotors, 2022-08-09:
The esc doesn't know anything about its load and there are no RPM limits. If the motor spins up to the the same rpm stable but fails with a load the most common culprit is an unstable Back-emf circuit on large high current designs. As for settings you can try changing the timing values but there is nothing I can change in software to fix hardware. I would suggest joining the slack, sharing the layout and schematic in hardware channel and see if we can figure out what is wrong. I hope you can understand there is nothing i can do or change in the firmware if all commercial targets have no issues.

AkunEverson, 2022-09-05:
hello,AlkaMotors！
&nbsp; &nbsp; &nbsp; &nbsp; Thank you for your warm help before. The last time the motor stopped running, it was the motor.Caused by the failure of the motor.
&nbsp; &nbsp; &nbsp; &nbsp; Your program, which has some questions about ADC current acquisition, I hope you can answer them.

Among them, the processing of current data is not very clear, as follows:
actual_current=(smoothed_raw_current * 3300/41)/(MILLIVOLT_PER_AMP)+CURRENT_OFFSET;
Among them, I don't understand the function of the number 41.&nbsp; 3300 is actually the upper threshold of the input voltage of 3.3V.MILLIVOLT_PER_AMPThis variable is the detection resistance value.

So, Is the upper red team not suitable for the following current detection principle?
R33=0.5mO

AlkaMotors, 2022-09-05:
The voltage the pin reads is 0 to 3300mV.
The range of the MCU ADC is 12 bit,  0 to 4096.  41 is just a rounding of 40.96. 

So for a quick current calculation.
If you have a current shunt of 0.5mO and your current sense amplifier is 20x. At 1A the op-amp is putting out 0.01V or 10mV. The MILLIVOLT_PER_AMP in TARGETS.H would need to be changed to 10.
If your current shunt is 0.001 and the amplification level is 50x then the millivolt per amp would be (0.001 x 1A x 50) = 50mv.

AkunEverson, 2022-09-07:
Thank you very much for your help. I have understood the content collected by ADC.

Now I'm going to use this program on a quadcopter, because this ESC program is compatible with many equipment functions, so I want to keep only the functions used by the aircraft, while others are not needed.Therefore, in the ESC EEPROM format, Those 48 bytes, how do I set them, which ones are turned off and which ones are turned on?

Appreciate your help to me!

closed 2022-11-22 by AlkaMotors