AlkaMotors/AM32-MultiRotor-ESC-firmware #89: Telemetry on pin PA14

Heclo, 2022-09-14:
I have a set of Razor32 V2 35A ESC's which I have converted to AM32 firmware, however the telemetry pad on these ESC's is connected to PA14 (usart2) and not PB6.

Is there some option for changing the telemetry TX pin to PA14?

AlkaMotors, 2022-09-20:
@Heclo Sorry i missed this. I think we can do a version output on pa14. The problem is, unless we have the exact hardware, chances are the current or voltage scale will be off so the output might be not accurate. What firmware file did the razor32 use?

Heclo, 2022-09-20:
Hi AlkaMotors, thanks for responding! The hardware layout is the C variant, I used the one for MP6531. The voltage reading must be on point, as the speedPID loop works like a charm, so does current limit and under voltage protection. I plan on designing my own PCB with hardware layout A, so if it is too much of a hassle to make this conversion, don't worry about it. ;)

AlkaMotors, 2022-09-21:
I added it quickly but i am unable to test myself until the weekend? You would also need to enable "30ms telemetry" in the config tool.
[RAZOR32_TEST_PA14_TLM.zip](https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/files/9620992/RAZOR32_TEST_PA14_TLM.zip)

closed 2022-11-22 by AlkaMotors