AlkaMotors/AM32-MultiRotor-ESC-firmware #101: 0-Low RPM Startup Torque for Battlebots Applications

Camo55, 2023-02-10:
I would like to add that in the Battlebots Scene, particularily in the smaller weight classes (150g, 1 lb, 3lb) This startup torque issue has plagued all speed controllers that aren't Simonk, which to this day has the best low rpm-high torque scheme at the 0-500 rpm range, which is the most critical for this application, notably because either people want to direct-drive a 30mm wheel with a 2205 1500kv motor on 3s, or for a majority of cases, spinning an 1806 or 1000kv 3035 or similar propdrive with a 3-10 oz 'flywheel' (Weapon) with 1000+ kg*mm^2 Inertia (My 1lb bot uses an emax rsII 2207 1600kv to drive a shell with 1400kgmm2 of inertia)

The problem being these are all sensorless motors and VESC is just too big to be viable, so we are forced to use what the quadrotor scene uses.

To some extent, setting startup torque to 150% and turning off stall protection helps, but it's still a big issue and most of the solutions seem to still be very inefficient, not to mention most of the smallest speed controllers are still BL-Heli_S

I would love to see an example of AM32 running on an emax bullet 30A.

AlkaMotors, 2023-02-11:
Hi @Camo55, The vehicle types it normally drives don't need to accelerate a large mass from low rpm. There are some things you can do to increase the power available at low rpm if your hardware can handle it. Are you on the slack or discord? what ESC do you have?

Camo55, 2023-02-12:
I am not on the discord yet, but that's ok.
I've set startup torque to 150%, timing to medium-low, and disabled stall protection to help with it, but there are still often jitters.
I use Emax bullet 30A bl-heliS and their 45A blheli32 race version

AlkaMotors, 2023-02-12:
How can I help? I would try an AM32 ESC ..if you find the startup torque lacking then let me know.

Camo55, 2023-02-16:
There seems to be very little documentation on various ways to interface with either flight controllers or individual esc's to flash the new bootloader to them, for example if I have an arduino nano, is it possible to flash simply with the signal and ground wires to the esc?

AlkaMotors, 2023-02-16:
The AM32 bootloader needs to be loaded on with a debugger tool like an st-link. After the bootloader is on then you can flash firmware and updates using a Betaflight flight controller or an Arduino. It uses the same protocol and tools as Blheli.

closed 2023-03-08 by AlkaMotors