AlkaMotors/AM32-MultiRotor-ESC-firmware #102: dalrc rorocket esc restart

llll2222, 2023-03-05:
dear friend，
   dalrc rocket 50a esc flash firmware fd6288-1.93 bootloder pa2 V11.bin all stm32 mcu  restart music  cycle can't control  motor.

AlkaMotors, 2023-03-05:
Are you sure that firmware file is the correct one for the dalrc rocket? It sounds like the wrong target is flashed. Can you connect with the config tool? If you can't try the pb4 bootloader.

llll2222, 2023-03-06:
thanks for you reply. I not sure that firmware for this esc board . this dalrc Rocket 50A esc board scheme use stm32f051k6u6 and fd62288. yes I flash bootloder pb4，so esc config tool can connect to esc .But start music cycle replay. 
[image]

AlkaMotors, 2023-03-08:
@llll2222 i'm going to close this issue. The fd6288 firmware is not the correct one for your board and its not in the supported hardware list. Please join the Discord or slack if you need help trying to find the correct one.

closed 2023-03-08 by AlkaMotors