AlkaMotors/AM32-MultiRotor-ESC-firmware #103: No Temperature Limit Celsius, and no Current Limit Amps

ariefadha, 2023-03-25:
Hi Folks,

Im not found any Temperature Limit Celsius, and no Current Limit Amps on  Input Tab. And can I set motor to free wheeling (without any break at all)?
[image]
[image]

AlkaMotors, 2023-03-25:
Use the latest config tool downloaded from the AM32 readme. Version 1.70 config tool is very old. Turn off "complementary PWM: for freewheeling.

ariefadha, 2023-03-25:
> Use the latest config tool downloaded from the AM32 readme. Version 1.70 config tool is very old. Turn off "complementary PWM: for freewheeling.

thanks, thats solved the problem, but with complementary PWM off, I can't get super slow movement (sinusoidal startup).