AlkaMotors/AM32-MultiRotor-ESC-firmware #104: Cannot change Running Brake Level in Multi ESC Config Tool 1.82

kaczork, 2023-03-25:
Hi! I'm using Rhinoesc 80A and I wanted to change Running Brake Level from default 10 to some lower value. Changing this option to whatever else than 10 just disables sinusoidal startup. ESC starts to just ignore this option. Am I doing something wrong or I do not understand something? I am a programmer, but before I dig into the code, maybe the solution is obvious and is my fault?

AlkaMotors, 2023-03-26:
Sinusoidal startup requires complementary pwm and maximum braking in order to function as its holding the rotor in a rotating field. Anything that reduces the braking function will make it not as strong and running brake level will reduce it a lot. Running brake level is meant more for propped air vehicles so reduce the motor braking effect when dropping from high rpm to low.

kaczork, 2023-03-26:
Okay, understood. My problem is as follows: I'm running car (crawler) with maximum speed (throttle 100%). Now I'm releasing throttle to 0% which causes the engine, whole drivetrain and wheels to just stops in place. I've thought I'll reduce running brake to make braking process more smooth... is there any way to make engine to increaset time needed to drop from higher RPM to lower? (of course except being more gentle in throttle operation ;))

nelsonconect, 2024-01-18:
Did you find a solution for this ?