AlkaMotors/AM32-MultiRotor-ESC-firmware #105: Rhinoesc and version 1.94

kaczork, 2023-03-25:
Hi! When flashing Rhinoesc 80A with version 1.94 (AM32_RHINO80A_F051_1.94.hex) it just makes beeps when started and just does not work at all. Flashing was done by Multi ESC Config Tool 1.82. I've tried so far:
1. setting default values in tool,
2. trying to calibrate throttle range (this is not possible at all).

Everyting seems to work when flashing back to 1.93. Am I doing something wrong?

AlkaMotors, 2023-03-25:
I'll check it out. Are you using a servo input?

kaczork, 2023-03-25:
I'm using a FlySky FS-BS6 receiver as input to drive ESC (if this is what you meant)...

AlkaMotors, 2023-03-26:
OK, i can cofirm the servo input was not working on 1.94. version 1.95 should fix this. 
https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/releases/tag/v1.95

kaczork, 2023-03-26:
Hi! Thank you for quick response! Hereby I confirm 1.95 fixes the startup issue. I don't know how it works in general, as I have just torn down my model, but it boots properly, seems the sinusoidal startup is there :)

closed 2023-03-26 by kaczork