AlkaMotors/AM32-MultiRotor-ESC-firmware #107: is there a list of cli commands for configurator to set them by cli?

docjones82, 2023-05-14:


AlkaMotors, 2023-05-14:
@docjones82 AM32 uses the same ESC protocol used by blheli_s or bluejay so it works with flight controllers or with usb directly. This is a bidirectional protocol with all the necessary safety checks and acknowledgements to set and read settings. What sort of serial device do you want to try and connect?

AlkaMotors, 2023-05-14:
Command line interface can mean a lot of different things. The betaflight cli commands for setting motor direction or 3d mode etc can be used. These are documented in the betaflight project. Otherwise we have move on from a cli interface many years ago.

docjones82, 2023-05-15:
was gonna try iterminal on iphone and a flight controller

AlkaMotors, 2023-05-15:
@docjones82 interesting.. what serial device are you planning on using?  Maybe I can help out with this. Are you going to use modified serial adapter or a flight controller to connect from the phone to the ESC?

docjones82, 2023-05-15:
will be a esp32 disquised as flight controller, will have passthru on it , i have it working with esc-config.com w computer to change settings works, flash does not, but thinking about just using it and bluetooth to esp and then use iterminal to send terminal commands and would love the help, would love to make your configurator work online ... gonna try putting up aws server and docker and see if it can be put up on webpage as app if so then i dont need to do any of this lol