AlkaMotors/AM32-MultiRotor-ESC-firmware #117: Request latest source for published binaries.

hydra, 2023-09-08:
Hi, not wanting to stir the pot, but under the terms of the GPL V3 license it's not permissible to release binaries without the source.

https://www.gnu.org/licenses/gpl-faq.en.html#ModifiedJustBinary

[image]

As I understand it, the 2.01 release, published here: https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/releases/tag/v2.01 is based on code in this repo which has the GPL V3 license and includes commits from contibutors and code copied from other GPL projects (Cleanflight/Betaflight).

https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/blob/master/LICENSE

If I'm mistaken, please accept my apologies, if not then under the terms of the GPL license I'm formally requesting that the code be published for all published released binaries.

❤️

hydra, 2023-09-08:
Since the release is made from this repo, please forgive me if the binaries are actually compiled from this repo: https://github.com/AlkaMotors/AM32_Arterytek

AlkaMotors, 2023-09-09:
I'm not sure what you are asking. What bf code is in these binaries?  What do you need to know?  the artery project is where those binaries came from and you will find those targets there.. I try and keep everything up to date but sometimes if i tweak a parameter or a timing, basically a user setting..  i will try it out in pre-release before doing a release.

hydra, 2023-09-10:
> I'm not sure what you are asking. What bf code is in these binaries? 

I looked though the commit history and found commits copied from BF, the commits from BF were relating to makefiles written by @nathensoi and copied into AM32 by @mikeller and merged by you.

> What do you need to know? the artery project is where those binaries came from and you will find those targets there.. I try and keep everything up to date but sometimes if i tweak a parameter or a timing, basically a user setting.. i will try it out in pre-release before doing a release.

Other contributors have contributed to this project knowing that it's GPL-3 and that when binaries are published or distributed, pre-release or otherwise, that the source for those binaries is made available.    You can see the contributors list here:  https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/graphs/contributors

Where's the source for the code for the GD32E230 binaries, searching this repo for E230 reveals no code (https://github.com/search?q=repo%3AAlkaMotors%2FAM32-MultiRotor-ESC-firmware%20E230&type=code) , yet the binaries are published on this repo with no link to where the source code can be found.

AlkaMotors, 2023-09-10:
Ah. That's right there is no repo for the gd32. We are not using any of that makefile work in the at32 or gd32 project . The one project that uses their makefile is the stm32 repo only and there was no 2.01 release of that project. None of the code from those commits will be in those binaries released.

closed 2023-09-10 by AlkaMotors