AlkaMotors/AM32-MultiRotor-ESC-firmware #124: The motor gets stuck when it slowly transitions from sine wave mode to normal rotation mode

TheFRLee, 2023-12-09:
This description is similar to the problem I encountered：

WIKI：
The KV of the motor is also a rating of the motors back emf. For example a 1000kv motor will produce twice the back emf than that of a 2000kv motor at the same rpm. As the esc accelerates and transitions from sine mode to normal driving operation it exits sine mode at a fixed rpm. For a 1000kv motor this will probably be high enough for the motor to continue operation depending on the divider ratio (see above) where the 2000kv might not and might continually restart in this region.
-------------------------------------------------------------------------------------------------------------------------------------------------------
Motor：1104 4100kv
Voltage：8.4v
paddle：No

How to solve or improve this problem?

This is my parameter adjustment interface：
[image]
[image]

TheFRLee, 2023-12-09:
/

TheFRLee, 2023-12-21:
If there is an option to set from sine wave mode to normal mode, the stroke folding ratio of normal mode is acceptable from 0-5%; 
that is, normal mode skips the throttle below 5%. 
Although there will be a speed jump when directly transitioning from sine wave to 5%, compared to stuck rotation, I think this jump to a certain speed is acceptable.

AlkaMotors, 2023-12-21:
@TheFRLee I would first lower pwm frequency to 24 kHz variable.  If one esc is working fine and the other is not I'm not sure what to suggest. If there is a physical problem with it adding more startup power is not an ideal way to solve that.