AlkaMotors/AM32-MultiRotor-ESC-firmware #128: Skystar AM60 not working

csicso, 2024-02-07:
Hi,
I've connected the ESC through an FC to configure it, and it was successful, I can update the firmware, etc, and everything looks good, but when I try to run the motors through the configurator, the motors just beep three times, in higher and higher tones... here a short video about it: https://www.youtube.com/shorts/Fbzw55OlzjA
Using 2x2 cell battery, 5000mAh, 120C, charged. Any idea what can be wrong? Thanks

norfeas, 2024-06-17:
Hello. Have you found a solution? I have the same problem with Skystar AM60.

csicso, 2024-06-18:
> Hello. Have you found a solution? I have the same problem with Skystar AM60.

Unfortunately no, switched to different esc