AlkaMotors/AM32-MultiRotor-ESC-firmware #131: Flashing Artery AT32F421 Bootloader

firstqyz8408, 2024-03-04:
When I follow the WIKI 
[https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/wiki/Flashing-Tekko32-45A-Single-ESC](url)
to flush the bootloader, I ran into the following problems:

1.Add the first program algorithm:
Load "D:\\WORK\\202402_AM32\\AT32F421_AM32_Bootloader-main\\Objects\\Am32.axf" 
No Algorithm found for: 08000000H - 0800250FH
Partial Erase Done (areas with no algorithms skipped!)
No Algorithm found for: 08000000H - 0800250FH
Partial Programming Done (areas with no algorithms skipped!)
Partial Verify OK (areas with no algorithms skipped!)
Application running ...
[image]

2.Add a second program algorithm:
Load "D:\\WORK\\202402_AM32\\AT32F421_AM32_Bootloader-main\\Objects\\Am32.axf" 
No Algorithm found for: 1FFFF800H - 1FFFF80BH
Partial Erase Done (areas with no algorithms skipped!)
No Algorithm found for: 1FFFF800H - 1FFFF80BH
Partial Programming Done (areas with no algorithms skipped!)
Partial Verify OK (areas with no algorithms skipped!)
Application running ...
[image]

Unsurprisingly, the ESC cannot be connected ESC Config Tool.

If I add both algorithms at the same time:
Load "D:\\WORK\\202402_AM32\\AT32F421_AM32_Bootloader-main\\Objects\\Am32.axf" 
Erase Done.
Programming Done.
Verify OK.
Application running ...

the ESC still cannot be connected ESC Config Tool.   I need help

AlkaMotors, 2024-03-07:
The first two steps you did are correct. What esc are you programming and what is the input pin that it uses for signal?

firstqyz8408, 2024-03-08:
Actually, I'm not sure. I connected the ESC Config Tool to update the firmware to AM32_WRAITH32_F421_1.93 before flush the bootloader. The motor can operate normally, so I think it should be PA2. But I have changed the pin, PA2 and PA4 have the same result.I'm using STLink and JLink.I also found an issue where the ARM.hex end address I compiled exceeded the start address of the APP.Is there a problem with my keil compiler version or setup

firstqyz8408, 2024-03-08:
> The first two steps you did are correct. What esc are you programming and what is the input pin that it uses for signal?

Sorry, It's my fault .The input pin is PB4,I directly use JFLAH flash the bootloader，the ESC Config Tool can be connected in some cases.Thank you for your recovery.