AlkaMotors/AM32-MultiRotor-ESC-firmware #136: Add S.Port and FBus Telemetry

jasc76, 2024-06-20:
Hi there
can you add the possibility to output S.Port and FBus telemetry? It is a really nice thing to buy opensource esc's and have telemetry directly from the esc and AM32 sadly doesn't have this

AlkaMotors, 2024-06-20:
Certainly.. additional telemetry protocols will be added. most like to the experimental branch.

parkflyer, 2024-06-29:
I would strongly support this! For many users FrSky telemetry was one of the main reasons to use BlHeli32…

Voglberg, 2024-07-07:
Das würde ich auch unbedingt unterstützen.

EmanuelStassar, 2024-12-15:
Here another FrSky user. I have several BLHeli32 ESC's and use the telemetry on all of them. So here is my +1 for adding it to the code. thank you some much! I can't code otherwise I would like to help out. I can test...