AlkaMotors/AM32-MultiRotor-ESC-firmware #157: skystar km65a AM32

YetiSPb, 2024-12-08:
Tell me, is it possible to install AM32 on SKYSTARS KM65A?

GrozaRobotovFPV, 2025-01-23:
Hi! I found its firmware in the v2.16 release on the website https://am32.ca/downloads "SKYSTARS_KM65_F421_2.16.hex". It seems that it is supported, but for some reason there is no AM32 in the file name, although others have it. I'm also wondering if it's supported.

GrozaRobotovFPV, 2025-01-23:
And I found it here https://github.com/am32-firmware/AM32/blob/797fc07ceebd4c9d688dab0b944e61452abda074/Inc/targets.h#L1017 (Target compatibility List)

AlkaMotors, 2025-02-05:
@GrozaRobotovFPV Sorry please use the new github repo here. This one is no longer checked or updated https://github.com/am32-firmware/AM32

closed 2025-02-05 by AlkaMotors